pkoehlers/maploader #1: MQTT topic/payload and map file save

chandranorth, 2022-01-18:
Thanks for this project, which is just what I was looking for!  Before I try and install it, I had a few questions:

1.  What is the MQTT topic and payload to change maps?
2. Is it possible to save a map?

Thanks in advance!

pkoehlers, 2022-01-18:
Hi, I have updated the readme to include the [MQTT topics](https://github.com/pkoehlers/maploader#mqtt-topics). I hope that clears things up.

This program saves the map on every map change. 

When you start the program, it assumes the current map is called "main" and if you send another name to the command topic, it will persist it under that name. That original map will then be restored when you switch back to the map called "main"

chandranorth, 2022-01-19:
Thanks for the quick response.  At the moment, I have a new Z10 with Valetudo just loaded, and it is working fine on a single floor.  I would now like to take it to a new floor...since it has not mapped this floor as yet, I would first like to save the current map, then have it map the second floor, and eventually save that as a second map ("2nd floor" in your example).  How would one go about doing this through MQTT?  Also, does it save the segment information for each map?

I am still learning about MQTT through Home Assistant, so  pardon my ignorance!

pkoehlers, 2022-01-19:
Segment information is saved together with the map. 

So in your case you can just switch to the "2ndfloor" map, carry your robot to the new area and start the cleaning. When you switch back to the "main" map everything should be in the same state as before.

If you want to be super safe with your current map data(shouldn't be necessary but this project does not seem to have many users...), you can back up your current map manually on the shell like this: `tar cvf /data/map-backup.tar /data/ri /data/map /data/DivideMap /data/config/ava/mult_map.json`

For Homeassistant you can read more about [MQTT Select](https://www.home-assistant.io/integrations/select.mqtt/) and use my config code from the readme. Then you can just switch between the floors by including the entity in lovelace and select the map you want to switch to.

chandranorth, 2022-01-19:
Thanks.  I'll try it out this weekend when I have more time.

chandranorth, 2022-01-22:
Just to let you know that I got it working!  Seems to switch fine between floors.  Thanks for your excellent coding!

pkoehlers, 2022-01-22:
Great to hear! You're welcome

Flipsoo, 2022-03-19:
Sorry for opening this again, but im kinda lost with the homeassistant part of the config... did i get it right, that i only need this select mqtt set in config after installing the software onto the robot?

in my configuration.yaml i have the following but it wont restart the robot if i set one option :/

`select:
  - platform: mqtt
    state_topic: "valetudo/maploader/map"
    command_topic: "valetudo/maploader/map/set"
    name: "vacuum_maploader_map"
    options:
      - "main"
      - "KG FT"
      - "EG FT"
      - "UG La"
      - "EG La"
      - "OG La"
      - "OG La 2"`

pkoehlers, 2022-06-05:
hi, sorry for the late reply, I reopened this issue as it goes into a similar direction...

first of all, the config looks fine to me. Best option would be to check the maploader logs or checking the MQTT topic with some other client to see what's going on. The maploader software on the robot should send a message to the state topic when starting.

Just one note: I haven't tested spaces in the floor names, so I'm not really sure if that might break something....

Your workflow basically seems right to me. First select the floor that you want to start mapping, then move the robot and let it create the map. 
When you first switch to an other map than main, the current map will be saved as "main". Then you can select e.g. KG_FT, move the robot to that floor and start cleaning.

g-work, 2022-06-16:
This isn't exactly a software question - but do you also have to move the docking station to the floor the robot is about to clean?

PS: Thank you for this great piece of software which exactly meets my setting (L10 Pro + Home Assistant + 4 floors)! Now I just need the courage and time to root the robot :)

pkoehlers, 2022-06-17:
The maploader itself does not really care, if there is a dock on a floor or not. I'm personally using it with 2 floors and only 1 dock.

Not having a dock on a floor causes some limitations in Valetudo/the firmware though:

- After cleaning, the robot will drive around to search for the dock (I cancel this with a HA automation)
- I think that you won't be able to create segments and other restrictions without a dock. Maybe a workaround would be to move the dock to each floor, create the segments, ... and then remove the dock afterwards? not tested though

PS: Let's open a new issue for each following discussion

closed 2022-06-17 by pkoehlers

g-work, 2022-06-21:
> PS: Let's open a new issue for each following discussion

Of course, I'm sorry. Just thought it would fit to the question concerning the workflow. I'll try to move the dock tomorrow.

PS: Perhaps it would be useful to use Github Discussions (like in valetudo) to discuss questions that are no issues?