pkoehlers/maploader #3: Unreliable map changes

pkoehlers, 2022-06-05:
Sometimes the maploader seems to successfully unpack the desired map files and backup the old map, but after the reload the map of the robot is empty.

I assume this is some syncing issue / conflict with some other process running on the robot.

Needs further investigation.

Jokuh, 2022-06-09:
What do you mean by "the map of the robot is empty"?
If you mean that Valetudo says: "no map data. that is resolved by starting a cleaning job on the robot. As [I was reading up on Valetudo](https://github.com/Hypfer/Valetudo/discussions/1291) that seems to be caused by the robot not broadcasting the map to valetudo.

pkoehlers, 2022-06-09:
That's what I thought at first too. 

But I actually logged into the robot, and saw in the logs that the maploader unpacked map with id e.g. 9, but the map actually had another ID and was really empty. Even after starting the full clean

pkoehlers, 2022-06-16:
Fixed with release 1.2.0

closed 2022-06-16 by pkoehlers