pkoehlers/maploader #4: No data showed in MQTT or /data/maploader

pwlgrzs, 2022-06-07:
Good day,
I have installed the binary following the instructions, but all I am getting is a log file with following content:
2022/06/06 11:06:17 Connected
2022/06/06 11:06:17 Subscribed to topic: valetudo/maploader/map/set
2022/06/06 11:06:17 Subscribed to topic: valetudo/maploader/map

There is nothing in the MQTT under valetudo/maploader/map
[image]

There is nothing but a log folder in /data/maploader.

Is this expected? Do I need to do something else?

pwlgrzs, 2022-06-09:
I pushed raw message "main" to valetudo/maploader/map/set and vacuum rebooted.
Went to the logs and here's what I see there:

2022/06/09 21:27:07 Received message: main from topic: valetudo/maploader/map/set
2022/06/09 21:27:07 Changing map from main to main
2022/06/09 21:27:07 saving current map
2022/06/09 21:27:07 data/ri/
data/ri/ri.json
data/ri/backupInfo.json
data/ri/mapra.json
data/ri/mapcleanset.json
data/ri/mapname.json
data/ri/ri_37.dat
data/map/
data/map/map_id.json
data/map/version_file.json
data/map/docker_pose.json
data/map/37/
data/map/37/fine.bin
data/map/37/docker_pose.json
data/map/37/coarse.bin
data/DivideMap/
data/DivideMap/37/
data/DivideMap/37/map_roi.json
data/DivideMap/37/doors_info.json
data/DivideMap/37/segmented_map.png
data/DivideMap/37/all_doors_info.json
data/DivideMap/37/contours_with_obstacle.png
data/DivideMap/37/room_neighbor_table.json
data/DivideMap/37/slam_map.png
data/DivideMap/37/segmented_map_with_obstacle.png
data/DivideMap/37/collision_map.png
data/config/ava/mult_map.json

2022/06/09 21:27:07 removing current map files
2022/06/09 21:27:07 restoring map from archive
2022/06/09 21:27:07 data/ri/
data/ri/ri.json
data/ri/backupInfo.json
data/ri/mapra.json
data/ri/mapcleanset.json
data/ri/mapname.json
data/ri/ri_37.dat
data/map/
data/map/map_id.json
data/map/version_file.json
data/map/docker_pose.json
data/map/37/
data/map/37/fine.bin
data/map/37/docker_pose.json
data/map/37/coarse.bin
data/DivideMap/
data/DivideMap/37/
data/DivideMap/37/map_roi.json
data/DivideMap/37/doors_info.json
data/DivideMap/37/segmented_map.png
data/DivideMap/37/all_doors_info.json
data/DivideMap/37/contours_with_obstacle.png
data/DivideMap/37/room_neighbor_table.json
data/DivideMap/37/slam_map.png
data/DivideMap/37/segmented_map_with_obstacle.png
data/DivideMap/37/collision_map.png
data/config/ava/mult_map.json

2022/06/09 21:27:07 map change complete, rebooting robot
2022/06/09 21:27:14 Received message: main from topic: valetudo/maploader/map

pkoehlers, 2022-06-09:
Looking good to me, you should now be able to find the topics in your MQTT
browser too. And if you use any other map name, you should get an empty map
after reboot

pwlgrzs, 2022-06-09:
That's what happened when I used a different name, yes. One thing I should have not done is doing change to a different map name while cleaning, robot froze, had to reset it.

MQTT looks like this now:
[image]

(after I sent "main" again).

pkoehlers, 2022-06-11:
yes, definitely a bad idea to change the map (especially in such a way), while the robot is cleaning....

closing this, as your issue/question is resolved

closed 2022-06-11 by pkoehlers

Flipsoo, 2022-06-29:
sorry for writing here again in an closed issue, but after setting up all mqtt entity correctly i can see that there is no `valetudo/maploader/#` traffic at all, so i tried to set `valetudo/maploader/map/set ` with payload `"main"` and i can see that this message is sent, but there is no answer or something from `valetudo/maploader/map/status`.

I have the file in /data und the _post_ script is adapted like your config states...

im so sorry for all that dumb questions - but this is such an nessecary feature for me:D

pkoehlers, 2022-06-29:
If you don't get messages in valetudo/maploader/#  , the maploader did not
start properly.

Please ensure that you did the chmod on the binary and check the logs in
/data/maploader/log/maploader.log

Flipsoo, 2022-07-01:
Thanks for your answer, i checked the `chmod`, checked the `/data/_root_postboot.sh` and everything is as noted in the readme but there is no folder named maploader created  - perhaps this could be due to the fact that i have an Dreame D9 and this simply wont start=(

pkoehlers, 2022-07-03:
Highly probable that D9 is armv7 architecture just like the F9.

Please try this version from the F9 support issue: https://github.com/pkoehlers/maploader/issues/8#issuecomment-1166261711

Otherwise, please post the output of calling these commands in a shell:
`uname -a`
`VALETUDO_CONFIG_PATH=/data/valetudo_config.json /data/maploader-binary`

Flipsoo, 2022-07-12:
Hey there, 
thanks for your reply. The `uname -a` shows 

`Linux p2009_release 4.9.118+ #1 SMP PREEMPT Tue May 25 15:43:41 CST 2021 armv7l GNU/Linux`

'VALETUDO_CONFIG_PATH=/data/valetudo_config.json /data/maploader-binary' doesnt show anything but im a bit sure, that there is a lack of understanding on my side - same for the version of maploader for the F9, im sorry but im not skilled enough to get it running.
I tried to change the weblink in the wget command, but doesnt works so i guess i have to build it somehow=)

pkoehlers, 2022-07-29:
Hi, I merged the mentioned PR, so it should be easier for you now.
Try the following:
```rm /data/maploader-binary```
```wget -O /tmp/maploader-binary https://github.com/pkoehlers/maploader/releases/download/v1.3.0/maploader-arm```
```chmod +x /data/maploader-binary```

Then reboot the robot (given you did the change in _root_postboot.sh) or execute the command manually again.