pkoehlers/maploader #8: Supporting the Dreame F9

JamesBelchamber, 2022-06-20:


JamesBelchamber, 2022-06-20:
The reason for this is likely because tar is failing - manually testing the tar attempt produces the following output:

```
[root@p2008_release:~]# tar zcvf test.tar.gz /data/ri /data/map /data/DivideMap /data/config/ava/mult_map.json 
tar: /data/ri: No such file or directory
tar: /data/map: No such file or directory
tar: removing leading '/' from member names
data/DivideMap/
data/config/ava/mult_map.json
tar: error exit delayed from previous errors
```

This gives an exit code of 1, matching the logs, and it's clear that some of the directory names are not the same on the Dreame F9.

(FWIW it's reasonable if you don't want to support this robot :) but if you do then I'm happy to hack on it a bit!)

pkoehlers, 2022-06-21:
hi, I'm open to supporting your robot, if you provide the input/testing.

For the start, have you identified the map files/folders? If unsure attach the output of something like `ls -lR /data`.
Also check for the existence of files `/etc/rc.d/miio.sh` and `/etc/rc.d/ava.sh`

JamesBelchamber, 2022-06-22:
Okay, so I've done an `ls -lR /data` before and after running a Map Reset from Valetudo. I see the following changes:
- `/data/DivideDebug/` has been completely cleared of all files and subdirectories.
- `/data/config/ava/mult_map.json has been updated.
- `/data/log/map_info.bin` (which contains a JSON object) has shrunk.
- `/data/log/ri/` has had all files updated:
  - `ri.json` (which contains a JSON object) has shrunk.
  - `ri_5.dat` has been replaced with `ri_0.dat` and has shrunk.
  - `mapname.json` and `mapra.json` seem to contain nothing.
- `/data/log/slam.db` (an sqlite database) has shrunk.

In addition, the following files have been updated - but seem to just be log files (actual log files) under `/data/log/` (these are likely irrelevant but provided for completeness):
- `fds.log`
- `script.log`
- `sys.log`
- `wifi_manager.log`

It's weird to me that `/data/DivideMap/` hasn't seen any updates - I'm running a clean and seeing if it gets updated at that point.

JamesBelchamber, 2022-06-22:
After a clean the following changes can be seen:
- `/data/DivideDebug` has been re-populated.
- `/data/DivideMap` contents have been changed (the only subdirectory, `5`, was replaced with the subdirectory `11` - with similar contents).
- `/data/config/ava` has had updates:
  - `clean_log.db`
  - `clean_log.json`
  - `clean_parameter.json`
  - `mult_map.json`
  - `vslam_extend.json`
- `/data/log/` got a bunch of updates:
  - `map_info.bin`
  - `slam.db`
- `/data/log/ri/` also got updates:
  - `mapname.json` is still empty but the timestamp has changed to `Jan 2 1970`.
  - `mapra.json` is still empty but the timestamp has changed to `Jan 2 1970`.
  - `ri.json` has been updated.
  - `ri_0.dat` has been replaced with `ri_11.dat`.

Again, the following log files were updated under `/data/log/` (probably irrelevant):
- `fds.log`
- `fds_log/` (this is a folder with a bunch of numbered, encoded files, which got an extra file).
- `miio_state.log`
- `msg_cvt.log`
- `script.log`
- `sys.log`
- `time.log`
- `wifi.log`
- `wifi_manager.log`

pkoehlers, 2022-06-25:
I compared your results to my robot and created a draft PR based on this.

Please check out the new ARM binary from the CI job here: https://github.com/pkoehlers/maploader/actions/runs/2560460085

JamesBelchamber, 2022-06-25:
It loads, it changes maps, it doesn't crash :sunglasses:

It's too late here now to test fully but tomorrow I'll run a full clean on two surfaces and then test switching between them. Looking good though :+1:

JamesBelchamber, 2022-06-28:
This seems to be working fine - the only thing is that it's not saving the second room, but I suspect that's because I am not setting up the dock in that room (so it can't "finish" the job). It switches back to the first room just as expected.

I would say this is ready to be merged now - I'll raise any issues I find as I find them. Thanks!

pkoehlers, 2022-07-03:
That's good to hear.

The saving of the second room actually works fine for me, even without placing the dock. Maybe you need to experiment a bit e.g. letting the robot search for the dock after cleaning vs stopping that process. But definitely not optimal without a dock....

JamesBelchamber, 2022-07-03:
I actually just put the dock in the room this weekend and kept running it
until it learned the room. Probably would've worked it out without the dock
but now it knows the layout it seems happy :)

Thanks again - this is definitely now in use! Feel free to tag me if you
ever need any testing on F9 hardware.