pkoehlers/maploader #11: Backup/restore

darsor, 2022-08-08:
This looks like a great project!

Is there currently any way to backup and restore a map? After a while the maps sometimes start to deteriorate and it would be nice to have a "master" backup to restore from in such cases.

darsor, 2022-08-08:
One solution that could achieve this would be to restore any given map name from /data/maploader via MQTT into the current map. Then one could manually copy/rename one of the automatic backups as a "master" backup, and restore it via MQTT.

pkoehlers, 2022-08-21:
Hi, good idea! That's actually something that I had to do manually a couple of times. 

I think it would be best to offer both, backup the current map with a backup name and restoring a named backup. What do you think?

Are you planning to work on the implementation too?

darsor, 2022-08-21:
Yeah, I'm working on an implementation of this (I've been looking for a reason to learn Go!). It basically splits the change map functionality into two parts: backup, then restore, either of which can be triggered individually. I'll try to wrap it up and get in a pull request this week.

I'm not sure what the best way to integrate those functions will be within home assistant, but I'll play with it a bit and see what works.

pkoehlers, 2022-08-23:
sounds good! I guess one option for HomeAssistant would be using https://www.home-assistant.io/docs/mqtt/service/ ,if HA integration is necessary at all.

closed 2023-08-20 by pkoehlers