pkoehlers/maploader #14: SAving the map without a dock

louis-lau, 2022-11-06:
Hi! How do you get the robot to save the map without having a dock on that map? I managed to get it working by cumbersomely moving the dock back and forth (The dock of the Z10 pro is quite big). But now it won't clean where it thinks the dock is, and the map will never be updated again. The robot also always goes to search for the dock that isn't there.

I know the default firmware can do this with the multimap feature, is there some way we can trick the firmware into thinking it's currently on a multimap without a dock?

pkoehlers, 2022-11-16:
Hi, thats something that somebody else have reported before. In my setup the robot saved the map without a dock on my second floor on the first try, so I did not really look into this further.

Did you have success in the meantime? maybe you could experiment with aborting the searching for dock process vs letting it search till the error occurs...

Using the multimap feature that the official app would probably be a superior method in general. But this will be some work to reverse engineer the calls and implement that

louis-lau, 2022-11-16:
I've experimented lots, but it just doesn't save the map unless it successfully docks after cleaning or mapping. Only a sort of temporary map that you can't place restrictions on. But if it's not the case for your robot it makes sense you didn't look into it.

g-work, 2022-11-16:
I experienced the same, but I don't see it as a big problem. The place where the dock was initially placed (and which is not cleaned now) is rather small and my walls don't change. Nevertheless it would be nice to have a solution.

louis-lau, 2022-11-16:
Yeah it's fine, it does work. Just wondered if someone else had this problem and maybe a solution 😁. But even without a solution I'm already quite happy.

pkoehlers, 2022-11-21:
I actually checked on my second floor and I cannot create segments or restrictions without the dock either. My second floor is just so small that I never even tried to do that...

pkoehlers, 2024-01-01:
For me this is working for quite some time now, by letting the vacuum create the map while the dock is (temporarily) present on each floor. 
With the added save / load methods it should be possible to load the correct map, in case the robot removes the dock from the map after cleaning

closed 2024-01-01 by pkoehlers