pkoehlers/maploader #17: Add option to not restart Valetudo

timoschwarzer, 2023-09-21:
Valetudo refreshes the map automatically at least every 60 seconds and map refreshing works fine for me without restarting Valetudo in the process. (Dreame W10)

pkoehlers, 2023-09-21:
Actually never tried without restarting. But will definitely check it on my
end.

Were you planning to submit a PR e.g. for an env var controlling this?

timoschwarzer, 2023-09-22:
I can make one later! I wasn't sure if I missed something and restarting it is actually required for some non-obvious reason 😅

pkoehlers, 2023-09-22:
Ok 😄 Only other reason for me to keep the Valetudo restart is that I use it in HomeAssistant to reliably tell, if a map change is completely done and segment cleaning can be started for the desired map

timoschwarzer, 2023-09-23:
@pkoehlers I added a new `starting_services` status that is set as long as the `ava` process is not ready yet. So *in theory* you could watch for a state change from `starting_services` → `idle` and *should* be able to start cleaning afterwards :sweat_smile:

AndreKR, 2023-10-13:
FWIW I did a cursory review and I am currently using this branch on my W10 and everything works as intended.

closed 2023-11-19 by pkoehlers

pkoehlers, 2023-11-19:
> @pkoehlers I added a new `starting_services` status that is set as long as the `ava` process is not ready yet. So _in theory_ you could watch for a state change from `starting_services` → `idle` and _should_ be able to start cleaning afterwards 😅
> 
> I didn't try automating it but I waited for it to be idle again and immediately started a zone cleaning, that worked.

Thanks for the PR! 

By the way, it does not directly accept a cleaning request, after state switches to idle. But this is still a bit faster than the Valetudo restart 👍

PR #18 (2023-11-19): Various improvements
- Makes restarting Valetudo optional
- Wait for robot to be active again. Status is set to `starting_services` during that time.
- Log is written to stdout and file
- General spelling/code style improvements

Closes #17