pkoehlers/maploader #19: Add L10s ultra

Aegeontis, 2023-10-08:
Hey, just installed this on a valetudo-rooted L10s ultra (arm64). Works fine, nothing seems to be out of order.

pkoehlers, 2023-11-19:
Hi, thanks for the report. 
When you start the maploader, it logs the profile that it detected as the best match and the model number for your robot. Can you share both?

Log lines should be:
```
Could not determine robot model using model: ...
Trying to identify compatible configuration. Please open a Issue/PR, if you think this configuration works well on your model
Using configuration for model:  ...
```

doemEINS, 2023-12-03:
Hey @pkoehlers,
just installed your maploader on my L10s Ultra. Here's the log you asked for:

```
[root@r2228_release:/data/maploader/log]# cat maploader.log
1970/01/01 00:00:07 Logging initialized
1970/01/01 00:00:07 Could not determine robot model using model: r2228
1970/01/01 00:00:07 Trying to identify compatible configuration. Please open a Issue/PR, if you think this configuration works well on your model
1970/01/01 00:00:07 Using configuration for model: p2029
1970/01/01 00:00:12 Connected
```

PR #23 (2023-12-03): feat: add supported model L10s Ultra
closes #19
closed 2023-12-03 by pkoehlers