pkoehlers/maploader #20: Allow custom mqtt topic

carefulcomputer, 2023-10-09:
If you can read valetudo config files and use valetudo topics, that will be be awesome. currently mqtt topics are hardcoded to valetudo/maploader/map/#

AndreKR, 2023-10-11:
This also means I can use maploader only on a single robot per MQTT broker, correct?

AndreKR, 2023-10-11:
Having `valetudo/maploader/map/` hardcoded also means I can use maploader only on a single robot per MQTT broker.

To solve that, we wouldn't need a fully custom topic, using `valetudo_config.json -> mqtt -> identity -> identifier` in the topic would be enough.

AndreKR, 2023-11-30:
@pkoehlers I saw you gave my comment a thumbs up, so I went ahead and created a PR: https://github.com/pkoehlers/maploader/pull/22

Since this changes the topics, it is a breaking change of course.

closed 2023-12-03 by pkoehlers