pkoehlers/maploader #21: Feature request : announce map change

carefulcomputer, 2023-10-10:
This is a feature request. When a map is changed, it would be awesome for vacuum to announce that new map has been loaded.

pkoehlers, 2023-11-19:
I'm not sure how that feature should look like.

In general the maploader offers a `Maploader state topic: valetudo/maploader/status`, which can be monitored for returning to status `idle`.

The new map should then be loaded, it may still take a moment until Valetudo is fully operational after the status change though.

tigattack, 2023-12-30:
The `aplay` binary can be used to play audio files, perhaps a bundled WAV could be played upon successful load of a new map.

This could also be implemented without direct support from maploader by triggering the aplay script from https://github.com/carefulcomputer/vacuumScripts upon a change to the `valetudo/maploader/status` MQTT topic.

PR #26 (2024-01-01): feat: audio playback (wav) on map change
closes #21
closed 2024-01-01 by pkoehlers