pkoehlers/maploader #24: Firmware-level multi-map functionality?

d235j, 2023-12-12:
Certain newer Dreame robot models have firmware functionality for storing and retrieving multiple maps. Some of the MIIO commands for manipulating maps are described in the README here: https://github.com/DeutscheKI/dreame-w10-valetudo-fork

However, it does not appear there's a known MIIO command to retrieve the list of the maps currently stored on the robot. It is possible that such a command exists, or it might be necessary to read the map list from the `/data/log/map_info.bin` or `/tmp/mult_map.bin` files on the robot.

A mechanism for handling multiple maps that uses these commands, as opposed to moving files around and forcibly restarting the robot software, would be preferred. However, more reverse engineering might be necessary here.

pkoehlers, 2023-12-31:
While I agree that the MIIO command alternative would be the preferred solution, I personally am not planning to put the necessary effort in that would be required for this.

Maybe it would even make sense to develop this as a separate project, especially if not all robots are supported?

Please feel free to build upon this code base or use this issue for discussions...

krusta4711, 2025-02-27:
I would be interested two. Especially as I do not find a way to mop on a floor without dock (see https://github.com/pkoehlers/maploader/issues/45).

I guess it would be a two-part investigation:

1. Find the needed MIIO commands
2. Observe which files are changed by the Dreame App when switching between maps. 

I'm not sure whether the analyzes is possible at all, as we most likely all have a FW without access to the Dreame cloud. So we cannot observe the original behavior.