pkoehlers/maploader #28: Help with initializing Dreame L10s Ultra

minnehahak, 2024-01-09:
I'm struggling to get any map loader response in MQTT from my Dreame (L10s Ultra) robot. Have tried for a number of hours so hoping you'd be able to help here. I have:

-Followed the installation directions carefully
-Checked that the correct file is in the correct location. I modified the permissions, and added the post boot instructions
-Updated today to 1.8 release (as well as latest Valetudo 2023.12.0), and cannot get a response both before or after updating.

When I call the service in Home Assistant (image attached), the MQTT Explorer does show the state change, but it doesn't update in the "select.vacuum_maploader_map" HA state. It seems so straightforward, I am unsure what to do from here or where I'm going wrong.

Thank you!

[image]

pkoehlers, 2024-01-09:
Is the status showing correctly in HomeAssistant?

can you double check / provide screenshots of the mqtt topics that are used? especially check the status topic to see if the maploader actually running at all

minnehahak, 2024-01-09:
Thank you for your help. I can call the map set function from Home Assistant and it shows up in MQTT explorer, but there is no response or publishing to the maploader topic from the robot. Screenshot of Explorer below after calling the service in HA.

One thing I noticed is that, under the valetudo category, a blank topic shows up with a separate maploader. You can see that when I called the "set" service in HA, the maploader nestled under my "Scrawny..." vacuum did change to the map (guestroom), but this blank child of valetudo is still there. If I delete it in MQTT explorer, it repopulates later.
[image]

Here is the state in HA after calling the service (it stays at "unknown")
[image]

Thank you!

pkoehlers, 2024-01-09:
That seems to be the problem then.

The blank part in the MQTT topic should be loaded from the valetudo config file, specifically:`mqtt.identitiy.identifier`. Can you check that the `VALETUDO_CONFIG_PATH` env variable is configured correctly and that the identifier is set in that file?

It might also make sense to check the log file at `/data/maploader/log/maploader.log` after a start of the maploader. It may contain errors loading the configuration

minnehahak, 2024-01-10:
Re the logs: The only thing that sticks out is “Connect lost: pingresp not received, disconnecting”, but there are not many errors printed.

SSH’d into the machine, `echo $VALETUDO_CONFIG_PATH` doesn’t return anything, so I must have missed that step in set up. Could you direct me to where I can find the valetudo config file and how to create that variable? Or, maybe I am just missing the config, because I see the `//` in the mqtt topic where the device name should be (see picture below).  Thank you.
[image]

AndreKR, 2024-01-10:
If you followed [these steps](https://github.com/pkoehlers/maploader/blob/main/README.md#steps) then you have added a line to `/data/_root_postboot.sh`. That line starts with `VALETUDO_CONFIG_PATH=/data/valetudo_config.json`.

minnehahak, 2024-01-13:
Thank you both for your help. I was able to get it working -- it seems that even though the vac identifier was correctly set in the env file, I had also manually filled it out in the Valetudo MQTT interface (at red arrow in the screenshot below), which I thought I had to do. I'm guessing that was causing some confusion, ending in the odd "//" situation we identified above. Once I deleted the identifier name I had added in the Valetudo settings, it worked. Thanks again and really appreciate this maploader!

[image]

closed 2024-01-13 by minnehahak

grilparto, 2025-01-08:
Sorry to resurrect a year old ticket, it can be closed right away but I have some context to add to this for anyone else encountering the same issue.

> That seems to be the problem then.
>
> The blank part in the MQTT topic should be loaded from the valetudo config file, specifically:`mqtt.identitiy.identifier`. Can you check that the [...]

I was having the same issue on my D10s Plus and noticed that the valetudo_config.json did not have the identifier called out under the mqtt config.  I also figured out why.  In my case, I was just using the autogenerated name for my vacuum provided by valetudo.  To get the "identifier" config to show up in the config file, I actually had to do the opposite of what @minnehahak did.  I filled in the topic prefix and identifier fields (they were already blank).  After doing that, rebooting the vacuum, and checking the valetudo_config.json file again, the "identifier" config is now included in the config file and everything seems to work properly.  Not sure if something has changed with the FW recently but short of manually adding the config to the file, the only solution was to set up an identifier in the Valetudo MQTT settings, otherwise that identifier either remains blank or not present in the config file.

AndreKR, 2025-01-08:
Yes, filling out the identifier is the way to go.

We can't (easily) fall back to the autogenerated name because that is not stored anywhere. Valetudo [generates it at runtime](https://github.com/Hypfer/Valetudo/blob/f795e2e6cc6c59265f22abd09998b5986803cf2d/backend/lib/utils/Tools.js#L128-L174). We'd have to replicate that code and keep it in sync if it changes in Valetudo. (Or ask @Hypfer if he could write it to a file for us.)

grilparto, 2025-01-08:
>  (Or ask @Hypfer if he could write it to a file for us.)

No no!  Filling out an identifier is simple and effective without needing anyone else to do anything 😁.  I mainly just wanted to comment on this in case anyone else ran into the same issue.  Oddly I did the opposite thing OP did to correct it.  Not sure how it'd even work the other way (given what I know now about how its generally working under the hood).

AndreKR, 2025-01-08:
Yeah, it shouldn't work without an identifier set. No idea what happened there.