pkoehlers/maploader #29: First time map name still "

JeroenEeuwes, 2024-01-26:
I just downloaded the binary on our L10S Ultra.

I ran it the first time using this command on the command line

 `VALETUDO_CONFIG_PATH=/data/valetudo_config.json DEAFULT_MAP_NAME=kitchen /data/maploader-binary > /dev/null 2>&1 &`

However it still said "main" in the MQTT map topic and not "kitchen". Do I not understand how I should've run the command the first time to get another map name as default? I now saved the map as "kitchen" by publishing in the map/set topic, so no man overboard. Just curious what I did wrong.

pkoehlers, 2024-01-26:
The way you passed the configuration looks fine, but you have misspelled
"DEFAULT"

JeroenEeuwes, 2024-01-26:
Right, that makes sense. I just need to learn how to type without autocorrect :|

closed 2024-01-26 by JeroenEeuwes