pkoehlers/maploader #30: D10S Pro

xXkr13g3rXx, 2024-01-28:
Is the D10S Pro supported? Would be nice to know before starting. If not known, I would try it.

pkoehlers, 2024-01-28:
It has not been added to the list of supported robots, but chances are good that it works out of the box with the 64bit binary.

In case it works, make sure to come back and share the detected configuration from the logfile `/data/maploader/log/maploader.log`, e.g.: `Using configuration for model: ...`

michelcve, 2024-02-23:
Did you try it? I have the same robot, so I'm curious as to whether it worked.

michelcve, 2024-02-25:
Okay, I'd be willing to try if noone else will. 

This is the logging I have so far after installing the binary:

```
1970/01/01 00:00:08 Logging initialized
1970/01/01 00:00:08 Could not determine robot model using model: r2250
1970/01/01 00:00:08 Trying to identify compatible configuration. Please open a Issue/PR, if you think this configuration works well on your model
1970/01/01 00:00:08 Using configuration for model: p2029
1970/01/01 00:00:13 Connected
```

These folders are present on the D10s Pro

```
    /data/ri
    /data/map
    /data/DivideMap
    /data/config/ava/mult_map.json
    /data/log/map_info.bin
```

` /data/log/slam.db`  was not present though.

After install and reboot, there is not other data than ` log`  in  `/var/maploader` . But that's probably because I haven't switched any maps yet.

Would be nice to have a bit more info on the homepage on how to add it to lovelace :-)

michelcve, 2024-02-25:
I switched map, and let it ' remap'  my second floor. After that I let it do a full clean.

So far so good, however, when I want to add a virtual wall, I receive an error that the map data was not persisted. And that I need to do a full clean first (which I already did).

Can this be related to that the docking station is NOT on the second floor?

michelcve, 2024-02-25:
Tried it another time WITH the docking station on the second floor, and this time the map is properly saved, and I can split sections.

Seems like the Dreame D10S Pro is working fine!

g-work, 2024-02-25:
> Can this be related to that the docking station is NOT on the second floor?
Yes, the full clean has to be finished at the dock to create a persistent map. I just put my dock on the second floor to finish the map and then put it back, you only need it once.

michelcve, 2024-03-01:
Yeah, that works great. You can add the Dreame D10S Pro to your list of supported robots.

And thanks for all the hard work, making this a reality. You made my wife very happy 👍  (and me too)

PR #33 (2024-03-10): feat: add supported model D10S Pro
closes #30
closed 2024-03-10 by pkoehlers