pkoehlers/maploader #31: Symlinks

michelcve, 2024-02-23:
Just curious, is there a reason why you are removing the map and untarring a new one upon map switching, instead of keeping an (untarred) version of the maps, and just symlinking them?

That might reduce the wear on the storage perhaps, and would be a bit faster?

carefulcomputer, 2024-03-05:
Another option is to use bind mount to mount different maps. I use bind mount for switching default voice prompts with my custom voice prompts

```
DIR="/data/customVoice/T2"
if [ -d "$DIR" ]; then
  # Take action if $DIR exists. #
  mount --bind ${DIR} /audio/EN
fi
```

michelcve, 2024-03-06:
That's a great idea too!

pkoehlers, 2024-03-10:
Actually I started with a simple tar command and added the gzip compression because the backups were taking too much space. 

So implementing this will definitely need more space and would limit the number of maps/backups to keep.

I'm also not sure if the storage wear really is significant, compared to the continuous map updates the robot is writing?

carefulcomputer, 2024-03-11:
most of the users will have 2 or 3 floor max. I am not sure if these will take up all the space. if not, symlinking/mount will be much faster. if maps do take up lot of space then it will be an issue. Maybe we can leave the space management to users (just like number of backups which could eat up all the space if it is higher.)

michelcve, 2024-03-15:
You got a point with the continues writing of the maps that to bot does anyway. I guess you are right and that it doesn't matter that much. Thanks.

closed 2024-03-15 by michelcve