pkoehlers/maploader #32: generate a unique event when map loading is finished loading and all services have restarted

carefulcomputer, 2024-03-03:
currently after loading the map the state goes to idle, if there was an event like 'map_ready' or something which indicates that vacuum is now ready with new map, that will be allow automations to work when new map is ready. (e.g. start vacumming when new map is fully loaded and ready to go)
Alternate is to check the sequence of events to trigger the automation but that is not that easy.

pkoehlers, 2024-03-10:
Which automation system are you using? 

I'd say doing this in HomeAssistant is actually easier than implementing this in the maploader (e.g. when to switch from map_ready to idle?)

I am using this configuration, where I enable the map selection controls when `sensor.vacuum_mapchange_status` is `ready`:

```
---
# mqtt
state_topic: valetudo/dreame-model/$state
name: "vacuum_valetudo_status"
```
```
---
# mqtt
state_topic: valetudo/dreame-model/maploader/status
name: "vacuum_maploader_status"
```
```
---
# template
sensor:
  - name: vacuum_mapchange_status
    state: >
      {% if is_state('sensor.vacuum_valetudo_status', 'lost') or is_state('sensor.vacuum_valetudo_status', 'disconnected') %}
        problem: valetudo
      {% elif is_state('sensor.vacuum_maploader_status', 'offline') %}
        problem: maploader
      {% elif is_state('sensor.vacuum_maploader_status', 'idle') and is_state('sensor.vacuum_valetudo_status', 'ready') %}
        ready
      {% else %}
        map_change
      {% endif %}
```

carefulcomputer, 2024-03-11:
thank you for prompt response ! so basically when `sensor.vacuum_mapchange_status` is `ready` that means new map has been loaded.

closed 2024-09-22 by carefulcomputer