pkoehlers/maploader #34: Question: How to set up a simple backup restore

M1scer, 2024-05-10:
Unfortunately, it is not entirely clear to me from the description how I can make a simple backup and restore.

My goal is to create a backup via HomeAssistant using a button, or to reload this backup using another button.

So I press Backup, then a backup is created from "main", but "main" is still used.

If I press Restore, "backup" is loaded and overwrites "main". The backup should remain as "backup" and the active map should be used as "main" again.

M1scer, 2024-05-10:
i want to share my HA scripts, they should work like described in the edit above. I can not confirm that this is working for 100% because my backup and main maps are identically atm.

But anyway, they are useful for someone else:

<details>
  <summary>Backup</summary>
  
  ```yaml
alias: "Valetudo: Backup Map"
sequence:
  - alias: Aktuelle Karte speichern
    choose:
      - conditions:
          - condition: state
            entity_id: sensor.vacuum_maploader_status
            state: idle
            for:
              hours: 0
              minutes: 0
              seconds: 0
        sequence:
          - service: select.select_option
            metadata: {}
            data:
              option: backup
            target:
              entity_id: select.vacuum_maploader_map_save
        alias: Speichere aktuelle Karte als "backup"
  - alias: Zurück zu "main" wechseln
    choose:
      - conditions: []
        sequence:
          - wait_for_trigger:
              - platform: state
                entity_id:
                  - sensor.vacuum_maploader_status
                to: idle
            continue_on_timeout: false
          - service: select.select_option
            metadata: {}
            data:
              option: main
            target:
              entity_id: select.vacuum_maploader_map_load
        alias: Lade die "main" Karte
mode: single
icon: mdi:map-marker-down
  ```
</details>
<details>
  <summary>Restore</summary>
  
  ```yaml
alias: "Valetudo: Restore Map"
sequence:
  - alias: Aktuelle Karte speichern
    choose:
      - conditions:
          - condition: state
            entity_id: sensor.vacuum_maploader_status
            state: idle
            for:
              hours: 0
              minutes: 0
              seconds: 0
        sequence:
          - service: select.select_option
            metadata: {}
            data:
              option: backup
            target:
              entity_id: select.vacuum_maploader_map_load
        alias: Wechsel zur Karte "backup"
  - alias: Zurück zu "main" wechseln
    choose:
      - conditions: []
        sequence:
          - wait_for_trigger:
              - platform: state
                entity_id:
                  - sensor.vacuum_maploader_status
                to: idle
            continue_on_timeout: false
          - service: select.select_option
            metadata: {}
            data:
              option: main
            target:
              entity_id: select.vacuum_maploader_map_save
        alias: Speichere aktuelle Karte als "main"
mode: single
icon: mdi:map-marker-up
  ```
</details>

closed 2024-05-10 by M1scer

Ulrar, 2024-11-28:
Awesome, thank you for documenting this, I was looking for the same thing.
I backed up my map now and tried a restore, it seems easy enough to just send an mqtt message manually if I ever need to restore it, that'll do.
I'll grab that tgz and store it somewhere off robot.