pkoehlers/maploader #38: Support Dreame X40

marcetad, 2024-07-23:
First of all, thanks for this great work! Valueable addition to the valetudo software. 

I've installed the maploader on the X40. I'm able to store/restore a map once. After succesfull restore the status changes to "offline". 

I've added the log output below.
I'm not sure if its related to the difference between configuration and the model number of the X40.

If I can help please let me know.

```

1970/01/01 00:00:08 Logging initialized
1970/01/01 00:00:08 Could not determine robot model using model: r2416
1970/01/01 00:00:08 Trying to identify compatible configuration. Please open a Issue/PR, if you think this configuration works well on your model
1970/01/01 00:00:08 Using configuration for model: p2029
1970/01/01 00:00:14 Connected
1970/01/01 00:00:15 Subscribed to topic: valetudo/stoffie/maploader/map
1970/01/01 00:00:15 Received message: main from topic: valetudo/stoffie/maploader/map
1970/01/01 00:00:15 Loaded current map from status topic
1970/01/01 00:00:15 Subscribed to topic: valetudo/stoffie/maploader/map/save
1970/01/01 00:00:15 Subscribed to topic: valetudo/stoffie/maploader/map/load
1970/01/01 00:00:15 Subscribed to topic: valetudo/stoffie/maploader/map/set
1970/01/01 00:00:15 Received message: main from topic: valetudo/stoffie/maploader/map
2024/07/23 09:22:53 Received message: "second_floor" from topic: valetudo/stoffie/maploader/map/load
2024/07/23 09:22:54 Attempting to load map "second_floor"
2024/07/23 09:22:54 stopping processes
2024/07/23 09:22:54 removing current map files
2024/07/23 09:22:54 loading map from archive
2024/07/23 09:22:54 data/ri/
data/ri/backupInfo.json
data/ri/3.dat2
data/ri/config.json2
data/ri/sub_trace_config.json
data/map/
data/map/map_id.json
data/map/version_file.json
data/map/3/
data/map/3/wifi_fine.bin
data/map/3/fine_large.bin
data/DivideMap/
data/config/ava/mult_map.json
2024/07/23 09:22:54 map load complete, syncing files
2024/07/23 09:22:54 restarting processes
2024/07/23 09:22:56 exit status 1

```

marcetad, 2024-07-24:
I've now added
 `	{model: "r2416", // Dreame X40
		MapFolders:       []string{"/data/ri", "/data/map", "/data/DivideMap", "/data/DivideDebug"},
		mapFiles:         []string{"/data/config/ava/mult_map.json", "/data/log/map_info.bin"},
		restartProcesses: []Process{MiioClientProcess, AvaProcess}},`

to robot_models, built and uploaded the binary to the robot. 
map change is succesfull, though still receive a "exit status 1"

pkoehlers, 2024-07-24:
Hi, actually I think that I am experiencing the same behaviour occasionally
on my L10s Ultra. I will look into it once I have time

closed 2024-07-26 by pkoehlers

pkoehlers, 2024-07-26:
can you please retest with the new release 1.8.3?

marcetad, 2024-07-26:
Done. Maploader doesn't stop after loading map.

This is the log output:
`2024/07/26 07:20:19 Attempting to load map second_floor
2024/07/26 07:20:19 stopping processes
2024/07/26 07:20:19 removing current map files
2024/07/26 07:20:19 loading map from archive
2024/07/26 07:20:19 data/ri/
data/ri/backupInfo.json
data/ri/3.dat2
data/ri/config.json2
data/ri/sub_trace_config.json
data/map/
data/map/map_id.json
data/map/version_file.json
data/map/3/
data/map/3/map_status.json
data/map/3/wifi_fine.bin
data/map/3/fine_large.bin
data/DivideMap/
data/DivideDebug/
data/config/ava/mult_map.json
data/log/map_info.bin

2024/07/26 07:20:19 map load complete, syncing files
2024/07/26 07:20:19 restarting processes
2024/07/26 07:20:21 exit status 1
2024/07/26 07:20:21 Command /usr/bin/avacmd msg_cvt {"type":"msgCvt", "cmd":"status_idle"} exited with code 1. Retrying in a second...
2024/07/26 07:20:22 exit status 1
2024/07/26 07:20:22 Command /usr/bin/avacmd msg_cvt {"type":"msgCvt", "cmd":"status_idle"} exited with code 1. Retrying in a second...
2024/07/26 07:20:28 Received message: main from topic: valetudo/cleaner/maploader/map
2024/07/26 07:20:28 Loaded current map from status topic
2024/07/26 07:20:28 Received message: second_floor from topic: valetudo/cleaner/maploader/map
2024/07/26 07:20:28 Loaded current map from status topic
`

krusta4711, 2025-02-23:
Small remark:
The X40 was not added to the README as supported robot.

klamathiker, 2025-06-15:
@marcetad Thank you very much! It also works for me with this version.

PR #39 (2024-07-26): X40 support & health check issue
Closes #38