pkoehlers/maploader #43: Maploader status is Offline after map changing

davids33, 2025-02-22:
I have a Dreame L40 Ultra with Valetudo 2025.01.0 in use.
So far I have been running firmware 1536. The map exchange worked perfectly. Now I have changed to firmware 1659 and since then the maploader no longer works properly. Switching from Main to Second_Floor works but then I get the status “Offline”. Then I can not switch the map back to Main. Fortunately, I could restore the map from the backup. 

The log says at the end:
2025/02/15 05:59:08 map load complete, syncing files
2025/02/15 05:59:09 restarting processes
2025/02/15 05:59:09 exit status 1

Is there anyone else with the same error?

pkoehlers, 2025-02-23:
please use this modified version of the maploader with increased logging and share the log to debug this issue:
https://github.com/pkoehlers/maploader/actions/runs/13481616610

krusta4711, 2025-02-23:
I have the same problem with the X40 with FW 1659. I installed your increased-logging version. Funnily with that version it does not exit anymore. Everything works.

Log with MapLoader v1.8.3 

```
1970/01/01 00:00:11 Logging initialized
1970/01/01 00:00:11 Using configuration for model: r2416
1970/01/01 00:00:16 MQTT connection failed: network Error : dial tcp 192.168.1.27:1883: connect: network is unreachable
1970/01/01 00:00:21 Connected
1970/01/01 00:00:22 Subscribed to topic: valetudo/dreame_x40/maploader/map
1970/01/01 00:00:22 Received message: main from topic: valetudo/dreame_x40/maploader/map
1970/01/01 00:00:22 Loaded current map from status topic
1970/01/01 00:00:22 Subscribed to topic: valetudo/dreame_x40/maploader/map/save
1970/01/01 00:00:22 Subscribed to topic: valetudo/dreame_x40/maploader/map/load
1970/01/01 00:00:22 Subscribed to topic: valetudo/dreame_x40/maploader/map/set
1970/01/01 00:00:22 Received message: main from topic: valetudo/dreame_x40/maploader/map
2025/02/23 10:20:09 Received message: OG from topic: valetudo/dreame_x40/maploader/map/set
2025/02/23 10:20:10 Saving current map as main
2025/02/23 10:20:11 data/ri/
data/ri/setting.json2
data/ri/2.dat2
data/ri/quick_info_record.json
data/ri/config.json2
data/ri/sub_trace_config.json
data/ri/0.dat2
data/ri/backupInfo.json
data/map/
data/map/2/
data/map/2/fine_large.bin
data/map/2/wifi_fine.bin
data/map/2/app_map.bin
data/map/2/docker_pose_large.json
data/map/2/map_status.json
data/map/map_id.json
data/map/version_file.json
data/DivideMap/
data/DivideMap/2/
data/DivideMap/2/last_seg_map_large.png
data/DivideMap/2/dec_map_roi_large.json
data/DivideMap/2/slam_map_ori.png
data/DivideMap/2/all_doors_info.json
data/DivideMap/2/empty_obs.txt
data/DivideMap/2/state_map.png
data/DivideMap/2/segmented_map_large.png
data/DivideMap/2/map_roi_large.json
data/DivideMap/2/user_confirm.txt
data/DivideMap/2/sneak_path.txt
data/DivideMap/2/segmented_map_color.png
data/DivideMap/2/origin_img_slam.png
data/DivideMap/2/ori_instance_label.txt
data/DivideMap/2/collision_map.png
data/DivideMap/2/ori_instance_sgm_map.png
data/DivideMap/2/laser_steps_info.json
data/DivideMap/2/doors_info_large.json
data/DivideMap/2/out_walls_large.json
data/DivideMap/2/hide_rooms_large.json
data/DivideMap/2/segmented_map_with_obstacle_large.png
data/DivideMap/2/dec_map_large.png
data/DivideMap/2/sneak_area.json
data/DivideMap/2/LDS_furniture_.txt
data/DivideMap/2/ori_slam_roi.json
data/DivideMap/2/unified_doors_info.json
data/DivideMap/2/state_map_ori.png
data/DivideMap/2/ori_slam_map.png
data/DivideMap/2/vision_doors_info.json
data/DivideMap/2/slam_map.png
data/DivideMap/2/contours_with_obstacle_large.png
data/DivideMap/2/carpet_map_large.json
data/DivideMap/2/room_neighbor_table_large.json
data/DivideMap/2/carpet_path_large.txt
[... 678 lines cut ...]
2025/02/23 10:30:46 stopping processes
2025/02/23 10:30:46 [DEBUG] Executing command: sh [/etc/rc.d/miio.sh stop]
2025/02/23 10:30:46 [INFO] Command succeeded: sh [/etc/rc.d/miio.sh stop] | Output: 
2025/02/23 10:30:46 [DEBUG] Executing command: killall [-9 ava]
2025/02/23 10:30:46 [INFO] Command succeeded: killall [-9 ava] | Output: 
2025/02/23 10:30:46 removing current map files
2025/02/23 10:30:46 requested map does not exist, loading blank map
2025/02/23 10:30:46 map load complete, syncing files
2025/02/23 10:30:46 [DEBUG] Executing command: sync []
2025/02/23 10:30:46 [INFO] Command succeeded: sync [] | Output: 
2025/02/23 10:30:46 restarting processes
2025/02/23 10:30:46 [DEBUG] Executing command: sh [/etc/rc.d/miio.sh]
2025/02/23 10:30:47 [INFO] Command succeeded: sh [/etc/rc.d/miio.sh] | Output: 
2025/02/23 10:30:47 [DEBUG] Executing command: sh [/etc/rc.d/ava.sh]
2025/02/23 10:30:48 [INFO] Command succeeded: sh [/etc/rc.d/ava.sh] | Output: killall: ava: no process killed
2025/02/23 10:30:48 After process restart
2025/02/23 10:30:49 After map publish
2025/02/23 10:30:50 Wait for processes
2025/02/23 10:30:53 before sound playback
2025/02/23 10:30:53 Received message: main from topic: valetudo/dreame_x40/maploader/map
2025/02/23 10:30:53 Loaded current map from status topic
2025/02/23 10:30:53 Received message: OG from topic: valetudo/dreame_x40/maploader/map
2025/02/23 10:30:53 Loaded current map from status topic

```

davids33, 2025-02-23:
First, thank you. With the modified version everything works fine. (Map change from Main to Second_floor and bank)
Please find the Log:

1970/01/01 01:00:09 Logging initialized
1970/01/01 01:00:09 Using configuration for model: r2416
1970/01/01 01:00:14 Connected
1970/01/01 01:00:15 Subscribed to topic: valetudo/DismalUnripeOkapi/maploader/map
1970/01/01 01:00:15 Received message: main from topic: valetudo/DismalUnripeOkapi/maploader/map
1970/01/01 01:00:15 Loaded current map from status topic
1970/01/01 01:00:15 Subscribed to topic: valetudo/DismalUnripeOkapi/maploader/map/save
1970/01/01 01:00:15 Subscribed to topic: valetudo/DismalUnripeOkapi/maploader/map/load
1970/01/01 01:00:15 Subscribed to topic: valetudo/DismalUnripeOkapi/maploader/map/set
1970/01/01 01:00:15 Received message: main from topic: valetudo/DismalUnripeOkapi/maploader/map
2025/02/23 20:29:22 Received message: second_floor from topic: valetudo/DismalUnripeOkapi/maploader/map/set
2025/02/23 20:29:23 Saving current map as main
2025/02/23 20:29:25 data/ri/
data/ri/sub_trace_config.json
data/ri/setting.json2
data/ri/3.dat2
data/ri/quick_info_record.json
data/ri/config.json2
data/ri/backupInfo.json
data/map/
data/map/map_id.json
data/map/version_file.json
data/map/3/
data/map/3/map_status.json
data/map/3/fine_large.bin
data/map/3/app_map.bin
data/map/3/wifi_fine.bin
data/map/3/docker_pose_large.json
data/DivideMap/
data/DivideMap/seg_result_info/
data/DivideMap/seg_result_info/labels.txt
data/DivideMap/seg_result_info/9.png
data/DivideMap/seg_result_info/scores.txt
data/DivideMap/seg_result_info/7.png
data/DivideMap/seg_result_info/0.png
data/DivideMap/seg_result_info/boxes.txt
data/DivideMap/seg_result_info/1.png
data/DivideMap/seg_result_info/3.png
data/DivideMap/seg_result_info/2.png
data/DivideMap/seg_result_info/5.png
data/DivideMap/seg_result_info/6.png
data/DivideMap/seg_result_info/4.png
data/DivideMap/seg_result_info/8.png
data/DivideMap/3/
data/DivideMap/3/origin_img_slam.png
data/DivideMap/3/user_confirm.txt
data/DivideMap/3/empty_obs.txt
data/DivideMap/3/out_walls_large.json
data/DivideMap/3/segmented_map_color.png
data/DivideMap/3/contours_with_obstacle_large.png
data/DivideMap/3/sneak_area.json
data/DivideMap/3/ori_slam_roi.json
data/DivideMap/3/segmented_map_large.png
data/DivideMap/3/unified_doors_info.json
data/DivideMap/3/all_doors_info.json
data/DivideMap/3/carpet_path_large.txt
data/DivideMap/3/map_roi_large.json
data/DivideMap/3/doors_info_large.json
data/DivideMap/3/vision_doors_info.json
data/DivideMap/3/slam_map_ori.png
data/DivideMap/3/ori_instance_label.txt
data/DivideMap/3/laser_steps_info.json
data/DivideMap/3/collision_map.png
data/DivideMap/3/hide_rooms_large.json
data/DivideMap/3/dec_map_roi_large.json
data/DivideMap/3/ori_instance_sgm_map.png
data/DivideMap/3/state_map.png
[... 738 lines cut ...]
data/DivideDebug/3/state_map.png
data/DivideDebug/3/slam_map.png
data/DivideDebug/3/segmented_map_with_obstacle_large.png
data/DivideDebug/3/room_neighbor_table_large.json
data/DivideDebug/3/carpet_map_large.json
data/DivideDebug/3/sneak_path.txt
data/config/ava/mult_map.json
data/log/map_info.bin

2025/02/23 20:30:06 map load complete, syncing files
2025/02/23 20:30:06 [DEBUG] Executing command: sync []
2025/02/23 20:30:06 [INFO] Command succeeded: sync [] | Output: 
2025/02/23 20:30:06 restarting processes
2025/02/23 20:30:06 [DEBUG] Executing command: sh [/etc/rc.d/miio.sh]
2025/02/23 20:30:07 [INFO] Command succeeded: sh [/etc/rc.d/miio.sh] | Output: 
2025/02/23 20:30:07 [DEBUG] Executing command: sh [/etc/rc.d/ava.sh]
2025/02/23 20:30:07 [ERROR] Command failed: sh [/etc/rc.d/ava.sh] | Error: exit status 1 | Output: killall: ava: no process killed
2025/02/23 20:30:07 After process restart
2025/02/23 20:30:08 After map publish
2025/02/23 20:30:09 Wait for processes
2025/02/23 20:30:13 before sound playback
2025/02/23 20:30:13 Received message: second_floor from topic: valetudo/DismalUnripeOkapi/maploader/map
2025/02/23 20:30:13 Loaded current map from status topic
2025/02/23 20:30:13 Received message: main from topic: valetudo/DismalUnripeOkapi/maploader/map
2025/02/23 20:30:13 Loaded current map from status topic

mkre, 2025-03-13:
I am seeing exactly the same behavior as @krusta4711 and @davids33 with my X40. Everything is working fine with the increased-logging version, but the default version is crashing after changing the map.

@pkoehlers, do you need any further information or help?

closed 2025-09-28 by pkoehlers