pkoehlers/maploader #45: Mopping not possible on floor without dock?

krusta4711, 2025-02-27:
Hi,

I have a x40. My dock is on the ground floor and am using MapLoader to also clean the second floor. That works fine with vacuuming. But somehow I do not find a way for mopping.

The problem:
The robot says "start cleaning" but the first thing it is doing is returning to dock. I tried different ways. 

1. Start a mopping downstairs, stops it and bring the robot to the second floor, load map, start cleaning
2. Drive the x40 out of dock manually
3. Take the x40 out of dock

Nothing works. Does it work out for you (and if yes: how)?

pkoehlers, 2025-03-01:
Hi,

on the L10S Valetudo offers the following Quirks (Robot -> Quirks) that need to be adjusted to prevent the robot from returning to dock:
- Mop Cleaning Frequency: every_segment (-> avoids returning during the cleanup)
- Pre-Wet Mops: Dry (this should solve your issue)

krusta4711, 2025-03-02:
Indeed, with the Dry configuration it works! Thank you. :-)

closed 2025-03-02 by krusta4711

RudolfRendier, 2026-03-15:
@krusta4711 

Can you elaborate on how you do this now?
I tried with pre-wet mops to 'Dry' and washing the mops before loading the map en taking the robot to the other floor.
But with 'Vacuum then Mop' the Robot wants to return to the dock after the vacuuming.