pkoehlers/maploader #46: Missing identifier value in valetudo_config.json

Adito5393, 2025-03-15:
On my valetudo version, the `valetudo_config.json` had the identifier value empty:
```yml
  "identity": {
    "identifier": ""
  }
  "customizations": {
    "topicPrefix": "",
    "provideMapData": true
  },
....
```
I figured it out by running it in console:
```bash
VALETUDO_CONFIG_PATH=/data/valetudo_config.json /data/maploader-binary
2025/03/15 16:44:47 Logging initialized
2025/03/15 16:44:47 Using configuration for model: r2228
2025/03/15 16:44:52 Connected
2025/03/15 16:44:52 Connect lost: EOF
2025/03/15 16:44:52 Connected
2025/03/15 16:44:53 Subscribed to topic: valetudo//maploader/map
2025/03/15 16:44:53 Received message: main from topic: valetudo//maploader/map
2025/03/15 16:44:53 Loaded current map from status topic
2025/03/15 16:44:53 Subscribed to topic: valetudo//maploader/map/save
2025/03/15 16:44:53 Received message: main from topic: valetudo//maploader/map
2025/03/15 16:44:53 Subscribed to topic: valetudo//maploader/map
```
I could set the `identifier` via the Valetudo WebUI (`/options/connectivity/mqtt`):

[image]

Now it all works as expected. Thanks for sharing your work!

Would you be interested in adding some extra info to the README? (I can make a PR if you want) As a troubleshoot section maybe?
Other related issues: https://github.com/pkoehlers/maploader/issues/28

nakermann1973, 2025-05-26:
FYI I had the same issue on two of my robots

pkoehlers, 2025-09-28:
Fixed by falling back to hostname of the device, if identifier is not set

closed 2025-09-28 by pkoehlers